Dlloydev/ESP32-ESP32S2-AnalogWrite #3: calling analogWrite() frequently causes the pin to spend extra time low

joshua-8, 2021-04-27:
@Dlloydev First of all, thank you for making a really cool library. Making analogWrite() supported is a big step towards making programs run on both esp32s and standard Arduino boards more easily!

I was looking at the output of the Fade example with an oscilloscope and noticed the signal wasn't quite what I would expect.
[image]
Each time analogWrite was called to change the duty cycle on the pin, it seems the pin goes low for approximately 200 microseconds. This interrupts the otherwise good 5kHz PWM signal and means that even if the signal is set to 100% there will be intervals where the pin is LOW after each time analogWrite is called.
This issue could cause unexpected results in any program where analogWrite is called frequently. 

Is ledcSetup getting called every time analogWrite is? That might be what's causing the issue because when I have used ledcSetup once then only ledcWrite later I don't get the dips.
Would it be possible to check if the channel, resolution, or frequency needs to be changed? If only the duty cycle needs to be changed then I don't think ledcSetup needs to be called.

Please let me know if I can help in any way.
Thanks,
    joshua-8

Dlloydev, 2021-04-27:
Thank you for the detailed feedback! Yes, I think you may have found the root cause ... I'll run some tests later today then hopefully prepare an update - if not today, then very soon ...
Regards,
D. Lloyd

Dlloydev, 2021-04-28:
The waveform is now looking good. Only update ledcSetup when necessary. All parameters are now checked if changes are made. With analogWrite running continuously in the loop, a 5kHz PWM signal looked good and consistent.

This is the waveform when the duty cycle is changed 25% to 50% repeatedly ....
[image]
This is the waveform when changing both duty cycle and frequency repeatedly ....
[image]

analogWrite is now updated to version 2.0.2.

Dlloydev, 2021-04-28:
Still some variations to test and another update needed.
Should be ready in 1 or 2 days.

Dlloydev, 2021-04-28:
Well, I think I've ironed out the last wrinkle ... updated to version 2.0.4

Dlloydev, 2021-05-02:
I believe this is now resolved, thanks again for the feedback!
Please re-open or start a new issue if there's any problems, suggestions etc.

closed 2021-05-02 by Dlloydev