Dlloydev/ESP32-ESP32S2-AnalogWrite #4: default resolution is different from normal Arduinos

joshua-8, 2021-05-02:
First of all, thank you so much for fixing the issue with ledcSetup getting called repeatedly! That issue is gone for me now, but I found another thing:
When trying to use this library as a direct replacement for the standard analogWrite function, I noticed that the output was never going above a few percent on.
Usually analogWrite has an 8 bit resolution, but this library seems to use 13 bits as default. It's really great that you provide the option to set higher resolutions since ESP32s can support that, but if it wouldn't be too difficult and if I'm not missing something, changing the default to 8 to be the same as the normal analogWrite would make this library even easier to use.
Thanks,
joshua-8

Dlloydev, 2021-05-02:
That's something I was considering but overlooked. Now the PWM defaults match the Arduino UNO.
Thanks

New version 2.0.6
- the default PWM settings are now 980 Hz with 8-bit resolution (same as Arduino UNO)
- added setPinsStatusDefaults() function to allow custom default settings  
- updated documentation

joshua-8, 2021-05-02:
Thank you so much for fixing it, everything looks good to me now!

closed 2021-05-02 by Dlloydev