Dlloydev/ESP32-ESP32S2-AnalogWrite #12: [QUESTION] Usage for LED fading

clau-bucur, 2021-10-03:
Thanks for this library!
My use case is fading up to 36 LED strips from an ESP32.

I want to ask about it's operation because I did not fully understand it from the description.
Taking for example an ESP32-S2 will it provide PWM capabilities for the pins: 1- 14, 21, 33-42, 45, so a total of 26 pins.
This much is clear.

What's not clear, for me, is: can I assign individual duty-cycles (PWM values) for each of these 26 pins so that I can fade each LED on it's own ? (the 8 PWM channels stuff gets me a bit confused)
Is there any way for this library to support more than 26 pins?

Dlloydev, 2021-10-03:
Using an ESP32, the library will work with 8 PWM pins having up to 8 unique frequencies and up to 8 unique duty cycles. This can be expanded to 16 if you use the LEDC functions for another 8 pins. Note that the limitation here is that you can then have 16 unique duty cycles on 16 pins, but there's still only 8 unique frequencies because there are only 8 timers available. The library uses channels 0,2,4,6,8,10,12,14. If using the LEDC functions to use additional PWM pins with unique duty cycles, you can use channels 1,3,5,7,9,11,13,15. The 8 unique frequencies would be shared as follows: 0-1, 2-3, 4-5, 6-7, 8-9, 10-11, 12-13, 14-15.

Additional reference:
- https://rntlab.com/question/esp32-pwm-frequency-selection/
- https://github.com/espressif/arduino-esp32/issues/1256

Using an ESP32-S2, the library will work with 8 PWM pins having up to 4 unique frequencies and up to 8 unique duty cycles because there are only 4 timers available.

> Is there any way for this library to support more than 26 pins?

Yes, in `analogWrite.h`, you'll need edit the `pinMask` constant ... line 19 for ESP32S2, line 26 for ESP32 where each enabled bit signifies a PWM pin.

clau-bucur, 2021-10-03:
Thank you for your answer. It's still not clear for me.

I am not interested in having different frequencies. One frequency is enough for all PWM pins.
Can I set individual PWM levels for each of the 26 pins on the ESP32-S2 (control the LED brightness separately for each pin)?

Dlloydev, 2021-10-03:
No, you can't get 26 independent PWM levels for the ESP32-S2. With the library, you can set up to 8 individual PWM levels and assign these to any 8 of the 26 PWM pins.

[Chip Series Comparison](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/chip-series-comparison.html?highlight=mcpwm)

clau-bucur, 2021-10-03:
Understood now. Thank you!

closed 2021-10-03 by clau-bucur