Dlloydev/ESP32-ESP32S2-AnalogWrite #13: Need to cycle power after an OTA update

nophead, 2021-10-04:
For a simple 8 bit 20KHz fan PWM on pin 22 of a DOIT ESP32 DEVKIT V1 :

In `setup()` I have:
```
  analogWrite(FAN, 0, 20000, 8); 
```
In `loop() `I have:
```
  fan_pwm =  ... // a function of temperature
  analogWrite(FAN, fan_pwm);
```

This works as expected but if I do an over the air update the PWM value gets stuck until I cycle the power and then it works again.

Do I need to do something in the OTA code to stop the PWM or reinitialise it?

Dlloydev, 2021-10-05:
I haven't used the [basic OTA update or web updater](https://lastminuteengineers.com/esp32-ota-updates-arduino-ide/) yet, but it looks interesting. Therefore, I don't have any suggestions at this time regarding what can be done in the OTA code.

Note that with this library, its not really necessary to configure the PWM pin in setup. It will self-configure the pin with whatever parameters you specify on the first iteration, then only apply anything that changes in subsequent loop iterations. Therefore, you could skip the setup and just use `analogWrite(FAN, fan_pwm, 20000, 8);` in the loop. Perhaps this would work.

nophead, 2021-10-05:
No that still doesn't work. I guess the variables in RAM all get cleared when the program restarts but perhaps the hardware doesn't, so things get out of step.

nophead, 2021-10-05:
Passing a value of -1 to detach the pin in ArduinoOTA.onStart() fixes the issue.

closed 2021-10-05 by nophead