Dlloydev/ESP32-ESP32S2-AnalogWrite #23: 'class Pwm' has no member named 'setServo'

uPesy, 2022-12-13:
Hi,
I'm trying to use the libray to control a servo, but it doesn't work when i'm using the `pwm.setServo` function. It's not in the `keywords.txt` file :/
[image]

Dlloydev, 2022-12-13:
Yes, thanks for reporting this. The error reported is correct ... I forgot to delete `setServo()` from the readme.

The servo's minimum, default and maximum microseconds are now set by using the `attach()` function. This is now similar to how other libraries work, however you also have a bit more control. The servo pin can auto-attach to the first free channel, or you can specify the specific channel to use. See `attach()` in the readme file.

closed 2022-12-14 by Dlloydev

uPesy, 2022-12-14:
> Yes, thanks for reporting this. The error reported is correct ... I forgot to delete `setServo()` from the readme.
>
> The servo's minimum, default and maximum microseconds are now set by using the [...]

Ok i understand better why it's doesn't work -> the doc was not up to date