Dlloydev/ESP32-ESP32S2-AnalogWrite #24: Can't compile for ESP32 DEV Board

jenshackel, 2023-01-13:
Hi. I've got trouble for compiling when using any other esp32 board instead of esp32s2.

I tried different versions of the library as well as arduino 2.03 / 1.18.19 and Visual Studio Code as well. I also used different versions of ESP32 lib (1.5.0, up to the latest 2.x)
All are producing the same failure message:

In member function 'uint8_t Pwm::tone(uint8_t, uint32_t, uint16_t, uint16_t)':
pwmWrite.cpp:91:65: error: 'lIn member function 'uint8_t Pwm::tone(uint8_t, uint32_t, uint16_t, uint16_t)':
pwmWrite.cpp:91:65: error: 'ledcChangeFrequency' was not declared in this scope
         ledcChangeFrequency(ch, frequency, config[ch].resolution);
                                                                 ^
edcChangeFrequency' was not declared in this scope
         ledcChangeFrequency(ch, frequency, config[ch].resolution);
                                                                 ^

In member function 'void Pwm::ledc_attach_with_invert(uint8_t, uint8_t, bool)':
pwmWrite.cpp:319:3: error: 'ledc_channel_config_t' has no non-static data member named 'flags'
   };
   ^
exit status 1
'ledcChangeFrequency' was not declared in this scope

It's really frustrating an I hope you've got a clue for me.
CHEERS!
Jens.

Dlloydev, 2023-01-13:
Actually, the latest required versions are:

* [ESP32 Arduino Version 2.0.6](https://github.com/espressif/arduino-esp32)
* [ESP32-ESP32S2-AnalogWrite 4.2.4](https://github.com/Dlloydev/ESP32-ESP32S2-AnalogWrite) (this repo)

The error message you've shown occurs for older versions of the ESP32 Arduino.

Good luck! 
David.

jenshackel, 2023-01-13:
Hi David,
Thanks for the fast reply. I'm going to test it in the evening and give a report later. Are you using IDE 2.0.x or 1.18?

Dlloydev, 2023-01-13:
I have both Arduino 1.8.19 and Arduino 2.0.3 on my PC.
It works with both, however I suggest starting with Arduino 1.8.19

jenshackel, 2023-01-13:
It can be so simple! Jusr uninstalled and re-installed IDE V1.8.19 and Lib V4.2.4 ... works instantly!
Nevertheless thank you very much for the exellent support. Now I can move on with my project and your fine library (exactly what I needed!)
best regards
Jens

closed 2023-01-13 by jenshackel