Dlloydev/ESP32-ESP32S2-AnalogWrite #29: Jittery servo sweep with ESP32-C3

nickrehm, 2023-03-06:
Servo sweep example leads to jittery servo response which also affects other added servos commanded to static position.
Also observed this behavior in a 50Hz and 10Hz loop with 6 servos--When any of the 6 was given a changing value in the loop (smooth sweep up/down, for example), all of the other servos jittered (5-10degrees) about their commanded static position. 

This was also observed with a single servo.

Possibly due to writeServo() calling attach() on every call and resetting something?

Dlloydev, 2023-03-06:
Thanks for reporting your observations.
I haven't tested servos with actual ESP32-C3 hardware, but I have a few examples in Wokwi using 6 servos that don't reveal this behavior. The attach() function only attaches to a pin once, then it just returns the attached channel number.
[ESP32_C3_6_Servo_Knob.ino](https://wokwi.com/projects/355852275661848577)
[ESP32_C3_6_Servo_Sweep.ino](https://wokwi.com/projects/358409595739559937)
Do you have an example I could take a look at?

Dlloydev, 2023-03-17:
I've completed a new Wokwi scope chip which showed the jitter and helped with resolving the issue. An update to the library (4.2.5) will be published shortly. 
Thanks for your feedback!

closed 2023-03-17 by Dlloydev