Dlloydev/ESP32-ESP32S2-AnalogWrite #33: Tone interval to a non-zero value makes buzzer never go off

factoidforrest, 2023-05-11:
Not sure why! Try setting it to a non zero value and perhaps it will do the same for you. 

I was just calling it in a simple loop on an esp32-c3

The c3 is somewhat uncommon so that may be the issue. 

Works fine otherwise!

Dlloydev, 2023-05-11:
The lowest frequency value for tone using an esp32-c3 appears to be 153 (when simulated on Wokwi).
Note that if the frequency value changes before the duration value expires, the duration is reset.
Therefore, the buzzer will be on continuously until the tone value remains constant for longer than the duration value.
[Example](https://wokwi.com/projects/364425827119911937)

Dlloydev, 2023-05-26:
Pushed an update that limits the minimum tone frequency to 153 when using the ESP32-C3.

closed 2023-05-26 by Dlloydev